kennethjiang/OctoPrint-Anywhere #5: Use plugin data folder for config file instead of base folder

foosel, 2017-07-03:
`self._basefolder` must not necessarily be writable by OctoPrint, it's your plugin's install location. You can rely on it being readable, but not writable, and it might also change on updates, effectively nuking your config.

OctoPrint also provides your plugin with a data folder you can retrieve via `self.get_plugin_datafolder()` which is what you should use instead - it's located in OctoPrint's general configuration folder, or rather in a sub folder specific to your plugin of that, should always be writable and also be in a good location for users to backup.

See also ["Injected Properties"](http://docs.octoprint.org/en/master/plugins/injectedproperties.html#injected-properties) and [`get_plugin_datafolder`](http://docs.octoprint.org/en/master/modules/plugin.html#octoprint.plugin.types.OctoPrintPlugin.get_plugin_data_folder).

kennethjiang, 2017-07-04:
good catch @foosel . will work on changing it now.

kennethjiang, 2017-07-04:
https://github.com/kennethjiang/OctoPrint-Anywhere/commit/8f27d34156bb0d4fe3222f5d1400612c45798090 and https://github.com/kennethjiang/OctoPrint-Anywhere/commit/9690f46178a240f86d2f1a13f8ed5e34e6a3b55c should fix it. Please review them @foosel

commit 9690f461 (2017-07-04): closes #5. silence any possible error in moving config file
closed 2017-07-04 by kennethjiang