kennethjiang/OctoPrint-Anywhere #9: Camera feed cannot be rotated or flipped

ChrisCrewdson, 2017-07-04:
My printer's camera is mounted upside down (like this https://www.thingiverse.com/thing:1646474). I have the "Flip webcam horizontally" and "Flip webcam vertically" options checked in OctoPrint.
If these settings can be read from the API, they should determine parameters sent to mjpg_streamer for rotation (or elsewhere as appropriate in the chain).

kennethjiang, 2017-07-05:
Thanks @ChrisCrewdson for proposing this feature. It's definitely on my roadmap. Will keep you posted when i get it implemented.

cwestpha, 2017-07-08:
Ideally, could it parse the config.yaml file and pull the webcam settings and replicate those? that way it will honor the flips, rotations, and other settings made in octoprint and use those.
http://docs.octoprint.org/en/master/configuration/config_yaml.html#webcam

kennethjiang, 2017-07-09:
Good point @ChrisCrewdson I was thinking about exactly that. However I ran into a few technical trickiness there. So I eventually implemented a simpler version, which requires you to set it in Anywhere dashboard. However you only need to do it once and it'll remember the setting.

Give it a swing by upgrading OctoPrint Anywhere to 0.2.2 and then heading to Anywhere dashboard. Let me know if you run into any problems.

ChrisCrewdson, 2017-07-11:
It looks like checking both of them is incompatible.
Having neither checked like this [image]
looks the same as having both checked like this
[image]
Both work great individually, but unfortunately I need both.

kennethjiang, 2017-07-11:
hmmm, what browser and OS are you using? how about their versions?

ChrisCrewdson, 2017-07-11:
This occurs on:
Ubuntu 17.04 with Chrome Version 59.0.3071.86 
Windows 10 with Chrome (latest version)

kennethjiang, 2017-07-12:
Thank you! Will work on it!

kennethjiang, 2017-07-13:
Just pushed a new release. Please take a look (you don't need to upgrade the plugin. Just log into https://www.getanywhere.io/app/devices ). Let me know if it still doesn't work.

ChrisCrewdson, 2017-07-13:
It doesn't look like it has changed. I tried all the combination with the same result as above.

kennethjiang, 2017-07-13:
can you force a refresh in your browser?

kennethjiang, 2017-07-13:
hold on, found a problem that may still causes a problem in your browser. fixing it now.

kennethjiang, 2017-07-13:
Now it should be fixed (for real this time). But I will leave this issue open until you confirm it @ChrisCrewdson

ChrisCrewdson, 2017-07-13:
It works! Thanks for all your attention and quick response on this.

closed 2017-07-13 by kennethjiang