kennethjiang/OctoPrint-Anywhere #10: Only showing one hotend

albinger, 2017-07-05:
My machine has 2 hotends.  Only one is showing up and not the one I actively use.

kennethjiang, 2017-07-05:
@albinger thank you for submitting this issue. I'm aware of this shortcoming of my plugin. However, my own 3D printer only has 1 hotend so it'll be tricky for me to test it even if I could code it up. Any chance you can collaborate with me on it?

albinger, 2017-07-05:
Certainly.

Let me know what to look for / send your way.

-Andrew

kennethjiang, 2017-07-05:
Can you email/google hangout me at kenneth dot jiang at gmail dot com ? I'd like to quickly chat with you to figure out what needs to happen to support dual hotends and how you can help.

kennethjiang, 2017-07-20:
@albinger Will you be able to work with me to look at how to extract data for the 2nd hotend?

kennethjiang, 2017-10-24:
@albinger This feature should be available now. Let me know if it works for you.

closed 2017-10-24 by kennethjiang