kennethjiang/OctoPrint-Anywhere #11: Despite installing it, it doesn't install

madmax2, 2017-07-05:
I click to install this plugin.
From the console message It looks like it is installing 

But when I open the settings it does not show 
and the plugins section shows it wasn't installed.

kennethjiang, 2017-07-05:
@madmax2 can you copy/paste the console message when you tried to install this plugin?

Also if you can ssh to your Raspberry Pi, can you run `pip list | grep OctoPrint-Anywhere` and let me know the output of it?

madmax2, 2017-07-05:
@kennethjiang 
Console message
https://pastebin.com/VfDAAhPY

I am not using raspberry pi, I installed octoprint on windows.
If the command you posted is to check if the folders of octoprint-anywhere exists, then it does show in explorer.

I can see there is 2 folders exists in C:\OctoPrint\venv\Lib\site-packages
-octoprint_anywhere
-OctoPrint_Anywhere-0.2.1-py2.7.egg-info

kennethjiang, 2017-07-06:
I have not tested my plugin on Windows and I guess that's why you ran into issues.

From the log file it seems that my plugin was indeed installed successfully. I'm 99% sure the reason why it doesn't show up in OctoPrint because of some errors during OctoPrint startup.

If you know how to get OctoPrint log file, please do so and send it to me  at kenneth dot jiang at gmail dot com . Otherwise I'll love to do a quick skype/google hangout chat with you to figure out what went wrong.

kennethjiang, 2017-07-19:
Closing it now due to lack of activities. Let me know if you still experience problems @madmax2

closed 2017-07-19 by kennethjiang