kennethjiang/OctoPrint-Anywhere #13: Temperature graph not shown with "Tempsgraph Plugin (0.2.2)"

chixxi, 2017-07-05:
After installing you plugin I did not see a temperature graph anymore in the user interface of octoprint. After uninstalling the plugin "Tempsgraph Plugin (0.2.2)" I see the graph again.

But without your plugin installed and only "Tempsgraph Plugin (0.2.2)" installed, it works fine. 

Could you check how this behaves in your case? Could that be related?

kennethjiang, 2017-07-05:
@chixxi I think it's very possible this issue is caused by some conflict between my plugin and Tempsgraph Plugin. I will look into it when I get a chance.

chixxi, 2017-07-05:
Thanks a lot for the quick feedback and let me know if you need further testing.

kennethjiang, 2017-07-20:
Hi @chixxi , sorry for the delay.

I installed Tempsgraph and looked into this issue. From Javascript console it appears to be a Javascript error from Tempsgraph plugin. Attached is the screenshot.

This error might have been caused the installation of Anywhere plugin, but without knowing how Tempsgraph works, trying to debug this issue for it is like finding a needle in a haystack for me.

I'd recommend you to submit an issue to Tempsgraph plugin developers. I'm sure they will be more efficient in finding out where the problem is.

[image]

kennethjiang, 2017-09-19:
@chixxi I'm closing this issue assuming that you have worked with Tempsgraph team to sort this out. However if you have reason to believe this is an issue caused by Anywhere plugin, please re-open it and provide additional info. Thanks!

closed 2017-09-19 by kennethjiang