kennethjiang/OctoPrint-Anywhere #14: Info / README and installation error "Could not import OctoPrint's setuptools"

unixb0y, 2017-07-05:
Hello,
First of all I would really appreciate a bit more infos, e.g. how to use and configure it after installation, so a README would be awesome 😊
Second, when trying to install, I get this error:
`Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?`
`Command python setup.py egg_info failed with error code 255 in /tmp/pip-64i4ibgi-build`
What should I do? I tried with pip and pip3, both failed.
Thank you in advance!

kennethjiang, 2017-07-05:
Good suggestion on README file. I will add it when things settle down a bit. The backend system is quite overwhelmed by all the users who have signed up in the past 2 days...

Did you use OctoPrint's plugin manager to install my plugin? I'm pretty sure if your Raspberry Pi was installed from the OctoPi image downloaded from OctoPrint website, you should only have python 2.7, not python 3 in your system.

unixb0y, 2017-07-05:
Thank you very much 👍 
The error was by me, I found out on the internet that it's best just using the Plugin manager 😊
I have python 3 because I installed it one day but looks like the Octoprint is like a different user than Pi or in a subdirectory so stuff is missing. You could change to that directory or just use the Plugin manager :)

kennethjiang, 2017-07-05:
Got it. Did you eventually install the plugin successfully?

unixb0y, 2017-07-05:
Yes I just installed it, now how do I set it up? 😆

unixb0y, 2017-07-05:
Oh there's a video! 
Would be cool to include it here, maybe under the Wiki page so you figure out how it works without having to install it first :)

closed 2017-07-20 by kennethjiang

szafranski, 2023-04-16:
For people that still have some issues - try this:
`export pip=~/oprint/bin/pip`