kennethjiang/OctoPrint-Anywhere #15: Feature request: Livestream / Camera view for "guests"

unixb0y, 2017-07-05:
I have a feature request:
Maybe with a "guest account" or so, friends can watch the Camera without having access to the "Pause" and "Stop" buttons?
That would be awesome! 
Thank you for your great work 👍 #

kennethjiang, 2017-07-05:
Yup this'd be a very useful feature! thank you for proposing that.

2 ways to implement it comes to my mind:
* Add a button to make the cam feed public. anyone who has the correct link can watch the cam feed. Very similar to google drive's function of **sharing as a link**.
* Invite your friends to sign up with their email. You can control what they can see/do on a 'settings' page.

Which one you think will work better for you?

unixb0y, 2017-07-05:
I would do it with sharing a link 😊
The user would have a link associated to his printer / his account and he can toggle it on / off :)
What do you think?

kennethjiang, 2017-09-19:
@unixb0y This feature has been implemented in latest version and it's working quite well. Please check it out.

closed 2017-09-19 by kennethjiang

unixb0y, 2017-09-19:
Thank you very much 😊
Already seen it, but both my printer and my Pi are currently not working / broken unfortunately :(
I'll try it as soon as they're back working again.. :)