kennethjiang/OctoPrint-Anywhere #16: Incorrect IP address displayed

sdgray, 2017-07-06:
I'm also using netconnectd plugin as a fallback for communication. OctoPrint Anywhere only displays the IP address that the host AP interface would have, if it were up, which would only be if no external connection were available.

ipOnConnect plugin still displays the correct IP address, always. Might want to check how that plugin finds IP address.

Most important would be the address of the interface with the default route.

kennethjiang, 2017-07-06:
Thank you so much @sdgray for pointing me to ipOnConnect plugin. I'll definitely check how it obtains the correct IP.

kennethjiang, 2017-07-20:
@sdgray do you have more than 1 wifi network interfaces on your Pi? Probably an internal one and a wifi dongle? I installed netconnectd but can't reproduce the same problem. 

A list of steps to reproduce this problem will be appreciated.

sdgray, 2017-07-20:
No, only the two interfaces total (ethernet and wifi).

kennethjiang, 2017-07-20:
Ah yes you are using ethernet too! 

I don't know if I want to use the way ipOnConnect detects IP address. As you have observed it only returns the IP of the default route. However I can see some use cases when this is not good enough. For instance, ethernet is attached to default gateway and the user wants to know the IP of Pi's wifi because the laptop is also on wifi network.

Ideally I should find out all routable IPs of OctoPrint and display them in OctoPrint Anywhere. But apparently there is a bug somewhere that prevents the code from doing it correctly.

Can you login to your Rasbperry Pi to run `hostname -I` and tell me if the output has all IP addresses, including the one of default route?

sdgray, 2017-07-20:
Here is all I can think to give you right now. Netconnect default AP IP
(10.250.250.1):

pi@Printer:~ $ hostname -I
10.1.1.141 10.250.250.1 169.254.223.207

pi@Printer:~ $ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use
Iface
0.0.0.0         10.1.1.1        0.0.0.0         UG    202    0        0 eth0
10.1.0.0        0.0.0.0         255.255.0.0     U     202    0        0 eth0
10.250.250.0    0.0.0.0         255.255.255.0   U     0      0        0
wlan0
169.254.0.0     0.0.0.0         255.255.0.0     U     303    0        0
wlan0
224.0.0.0       0.0.0.0         240.0.0.0       U     0      0        0
wlan0
239.255.255.250 0.0.0.0         255.255.255.255 UH    0      0        0
wlan0

pi@Printer:~ $ ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:2e:25:6f
          inet addr:10.1.1.141  Bcast:10.1.255.255  Mask:255.255.0.0
          inet6 addr: fe80::5585:b334:3bdf:2db0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1220213 errors:0 dropped:52 overruns:0 frame:0
          TX packets:1705117 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:504512182 (481.1 MiB)  TX bytes:2058740331 (1.9 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:309973 errors:0 dropped:0 overruns:0 frame:0
          TX packets:309973 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1936051712 (1.8 GiB)  TX bytes:1936051712 (1.8 GiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:7b:70:3a
          inet addr:10.250.250.1  Bcast:10.250.250.255  Mask:255.255.255.0
          inet6 addr: fe80::fb8b:18a3:4359:bf4d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:384817 errors:0 dropped:134366 overruns:0 frame:0
          TX packets:367877 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:51296078 (48.9 MiB)  TX bytes:471011424 (449.1 MiB)

pi@Printer:~ $ ip route
default via 10.1.1.1 dev eth0  metric 202
10.1.0.0/16 dev eth0  proto kernel  scope link  src 10.1.1.141  metric 202
10.250.250.0/24 dev wlan0  proto kernel  scope link  src 10.250.250.1
169.254.0.0/16 dev wlan0  proto kernel  scope link  src 169.254.223.207
 metric 303
224.0.0.0/4 dev wlan0  scope link
239.255.255.250 dev wlan0  scope link

kennethjiang, 2017-07-20:
Thank you @sdgray for the detailed information. I just released a new version that hopefully will correctly obtain and display all these ip addresses. Please upgrade and let me know if it actually works (I don't have an ethernet cable or router so I can't test it myself).

sdgray, 2017-07-20:
That works just fine. All IPs listed!

kennethjiang, 2017-07-20:
Hooray! Thank you so much @sdgray for reporting and helping me with this bug!

closed 2017-07-20 by kennethjiang