kennethjiang/OctoPrint-Anywhere #18: This plugin crashes the telegram plugin

derpicknicker1, 2017-07-09:
I had to support two users with not working telegram plugis. Both got it back to work after deactivating OctoPrint-Anywhere. Here are some of there errors/warnings

```2017-07-09 22:45:47,717 - backoff - ERROR - Backing off __probe_auth_token__(...) for 2.3s (HTTPError: 401 Client Error: Unauthorized)
2017-07-09 22:45:47,727 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.20
2017-07-09 22:45:48,291 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-07-09 22:45:48,885 - octoprint.server.api.settings - WARNING - Could not load settings for plugin OctoPrint Anywhere (0.2.1) since it called super(...)
2017-07-09 22:45:48,897 - octoprint.server.api.settings - WARNING - in a way which has issues due to OctoPrint's dynamic reloading after plugin operations.
2017-07-09 22:45:48,897 - octoprint.server.api.settings - WARNING - Please contact the plugin's author and ask to update the plugin to use a direct call like
2017-07-09 22:45:48,928 - octoprint.server.api.settings - WARNING - octoprint.plugin.SettingsPlugin.on_settings_load(self) instead.
2017-07-09 22:45:49,411 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-07-09 22:45:49,942 - octoprint.server.api.settings - WARNING - Could not load settings for plugin OctoPrint Anywhere (0.2.1) since it called super(...)
2017-07-09 22:45:49,968 - octoprint.server.api.settings - WARNING - in a way which has issues due to OctoPrint's dynamic reloading after plugin operations.
2017-07-09 22:45:49,986 - octoprint.server.api.settings - WARNING - Please contact the plugin's author and ask to update the plugin to use a direct call like
2017-07-09 22:45:50,020 - octoprint.server.api.settings - WARNING - octoprint.plugin.SettingsPlugin.on_settings_load(self) instead.
2017-07-09 22:45:50,465 - octoprint.server.api.settings - WARNING - Could not load settings for plugin OctoPrint Anywhere (0.2.1) since it called super(...)
2017-07-09 22:45:50,466 - octoprint.server.api.settings - WARNING - in a way which has issues due to OctoPrint's dynamic reloading after plugin operations.
2017-07-09 22:45:50,467 - octoprint.server.api.settings - WARNING - Please contact the plugin's author and ask to update the plugin to use a direct call like
2017-07-09 22:45:50,467 - octoprint.server.api.settings - WARNING - octoprint.plugin.SettingsPlugin.on_settings_load(self) instead.
2017-07-09 22:45:50,622 - backoff - ERROR - Backing off __probe_auth_token__(...) for 2.3s (HTTPError: 401 Client Error: Unauthorized)
2017-07-09 22:45:53,633 - backoff - ERROR - Backing off __probe_auth_token__(...) for 1.2s (HTTPError: 401 Client Error: Unauthorized)```

Niklas420, 2017-07-09:
I am one of the Users after uninstalling your Plugin the other one worked perfect. The Main Issue was that when i connected the Telegram OctoPrint Plugin with my phone the Chat won't appear in OctoPi.
Even when i waited for 10 minutes.

kennethjiang, 2017-07-09:
Sorry that my plugin broke yours @derpicknicker1 

From the log it's not obvious what is causing the problem (the 401 is expected before users link their printer to Anywhere). But I will look into it whenever I got a chance.

kennethjiang, 2017-07-09:
Also can you tell me how your plugin is broken @derpicknicker1 ? I don't have a telegram account so I'm not familiar how your plugin works.

derpicknicker1, 2017-07-10:
I will do some investigation the next days. I don't know exactly what the issues are, because i am not using your plugin 😉 but i will install it and have a look when i have time.
BTW: Your slicer plugin is very great!

kennethjiang, 2017-07-10:
Thank you @derpicknicker1 

I just installed Telegram. And I was able to see how Telegram settings page is broken after Anywhere is installed. Since my plugin causes warning messages about settings, it is very likely that it broke some part of settings in Telegram.

I will dig deeper to see how I can fix the warning message for my plugin, and see if Telegram will be restored after I fix it. Again my apology for the problem.

HogensHero, 2017-07-10:
I was the other user helped and was able to get a good log. Don't know if it will help but here it is. https://pastebin.com/0ErTBTBL

kennethjiang, 2017-07-10:
Thanks @HogensHero !

kennethjiang, 2017-07-11:
@derpicknicker1 I just released 0.2.4 that eliminated the warnings about settings. However, I tested your plugin and the settings page is still broken. Any insights why this could be the case?

kennethjiang, 2017-07-14:
@derpicknicker1 Does this issue still exist? Or you managed to solve it yourself?

kennethjiang, 2017-08-06:
Assuming this is no longer an issue I'm closing it. @derpicknicker1 feel free to reopen it if the problem is still there.

closed 2017-08-06 by kennethjiang