kennethjiang/OctoPrint-Anywhere #19: Reduce time between snapshots <Enhancement>

modem7, 2017-07-14:
Heya, 

It seems that the plugin updates every ~5 seconds between camera snapshots. Is it possible to have a choice of the refresh duration (e.g. a dropdown or input box)?

kennethjiang, 2017-07-14:
Yes it's on the roadmap!

kennethjiang, 2017-07-14:
One question though: do you want the refresh rate to be adjustable, or you just want higher refresh rate? The latter is easier to implement, but will cost higher data rate (it may or may not matter for the people with limited mobile data plan)

kennethjiang, 2017-08-06:
@modem7 I increased the frame rate to 5Hz but then the backend service was overwhelmed. Now I temporarily reduced the frame rate to 0.5Hz. After I optimized the backend service, I will bump it up again.

i also implemented a toggle on dashboard so that you can switch on "bandwidth saving mode".

closed 2017-08-16 by kennethjiang