kennethjiang/OctoPrint-Anywhere #20: Can't delete or rename printers

dustin, 2017-07-16:
I've got two printers, and had replaced the server servicing one of them at one point, so my dashboard currently shows three printers named "My First Printer" where one of them is never going to be online again.

kennethjiang, 2017-07-18:
It's definitely on my roadmap. I'm tied up with other tasks recently. But will push a release soon in the near future that includes this one. Stay tuned.

kennethjiang, 2017-07-19:
@dustin Your request has been implemented and released. It is a server-side change so you just come to https://www.getanywhere.io/app/devices# and refresh the page. No plugin upgrade is needed!

Let me know if you experience any problems.

closed 2017-07-19 by kennethjiang