kennethjiang/OctoPrint-Anywhere #22: Can't Create Printer

Sabreur, 2017-07-25:
I deleted my old printer setup, but now there's no option to create a new printer.  The dashboard just takes me to an empty page with "OctoPrint Anywhere" at the top.  I can open the left-hand menu, but the only options are "Dashboard", "Help", "Log Out" and "Report A Bug".  Clicking Dashboard just takes me back to the same empty page.

kennethjiang, 2017-07-25:
You can re-add your printer by following the instructions here: http://www.getanywhere.io/pub/help#missing-printer

Let me know if you still can't see the printer after going through these steps. @Sabreur

kennethjiang, 2017-08-06:
@Sabreur feel free to re-open this issue if you still run into this problem.

closed 2017-08-06 by kennethjiang