kennethjiang/OctoPrint-Anywhere #24: Huge issue

karatetrigger, 2017-07-29:
Excuse me, but I have a major problem with the screenshots you have on the Octoprint page. For some reason, they are of my 3D printer. Now, how in the hell did you get those images, and who do you think you are?

kennethjiang, 2017-07-31:
@karatetrigger Responded to you in another issue you opened. Like I said I will fix that by end of day tomorrow. Again my apology for using it without your permission.

kennethjiang, 2017-08-01:
I have updated all content, including pics and videos, to remove any images of your printer. @karatetrigger 

I don't think I missed any but let me know in case I did.

My apology again for using your pics without your permission!

closed 2017-08-01 by kennethjiang