kennethjiang/OctoPrint-Anywhere #25: Printer never connects

TeaVirus, 2017-07-29:
I'm able to "pair" the printer from Octoprint and it shows up in the dashboard.  The problem is that it continually shows "Waiting for camera feed..." and never gets any further.  Any recommendations?

kennethjiang, 2017-07-31:
Hi @TeaVirus The service that my plugin is using has started to throttle the requests because of the unexpected amount of signups to my plugin. I'm working with them to solve this issue. I will notify you once I get this figured out, probably in a few days to a week. If it still doesn't work after that, I will do some further diagnosis to see what could be the cause.

kennethjiang, 2017-08-06:
@TeaVirus I have solved the issues with our backend service. For the past several days it has been very stable.

I'm closing this issue now but feel free to reopen it if you still run into problems.

nkuck, 2019-01-04:
I am having the same issue. Is there a fix? Thanks for the help.

kennethjiang, 2019-01-04:
I suspect your Pi doesn't have a reliable internet connection. Can you send me `octoprint.log` so that I can take a look? @nkuck

nkuck, 2019-01-04:
Thanks, but got it working a couple of hours ago. Registered a new printer and deleted the old one.

kennethjiang, 2019-01-04:
Glad to know it worked for you.

closed 2019-01-04 by kennethjiang