kennethjiang/OctoPrint-Anywhere #26: Install fails with memory error

theisgroup, 2017-08-04:
Trying to install anywhere on fresh octoprint .14 on rpi zero wireless

kennethjiang, 2017-08-04:
Another user has reported a similar problem. He found out that the problem seems to be caused by pip version 9.0.1.

If you have upgraded pip to 9.0.1, you can downgrade it to 8.1.1 by the command `pip install pip==8.1.1`. Then you can try to install OctoPrint Anywhere again. Let me know if it solves your problem @theisgroup

theisgroup, 2017-08-05:
here is what i get after i downgrade to pip 8.1.1

Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

theisgroup, 2017-08-05:
i ssh into the pi and ran the install and it looks like that worked. i will check it once this print is complete.

kennethjiang, 2017-08-05:
Great. Meanwhile I will try to reproduce the problem on my Pi zero to see if I can come up with a permanent solution.

kennethjiang, 2017-08-06:
@theisgroup I released a new version yesterday so that the installation would succeed even under pip 9.0.1.

closed 2017-08-06 by kennethjiang

theisgroup, 2017-08-06:
i just flashed a second pi zero w with octoprint 0.14 and the plugin installed from plugin manager.