kennethjiang/OctoPrint-Anywhere #28: File manager plugin is broken by this plugin

kennethjiang, 2017-08-10:


astateofblank, 2017-11-02:
I experienced an issue with the Filament Manager as well that may be related. In:

https://github.com/kennethjiang/OctoPrint-Anywhere/blob/master/octoprint_anywhere/static/js/anywhere.js

`'use strict';` is declared globally. When OctoPrint bundles all the plugins' JavaScript, all other plugins must then be compliant with strict mode. Given the state of OctoPrint's plugin ecosystem, the entire script should be wrapped in an anonymous function to prevent conflicts (per the template). `'use strict';` should be declared within that function. Ideally OctoPrint would do this when bundling, and I'm submitting an issue there as well.

kennethjiang, 2017-11-02:
Hah this makes a lot of sense. Thank you so much @astateofblank ! I will make the change to see if it solve the problem.

kennethjiang, 2017-12-06:
Hi @astateofblank I just pushed out version 0.6.0 that uses IIFE to keep `use strict` within the scope of Anywhere plugin. The conflicts with FileManager and potentially other plugins should be eliminated now. Please let me know if you are still seeing problems.

closed 2017-12-06 by kennethjiang