kennethjiang/OctoPrint-Anywhere #29: No webcam feed

modem7, 2017-08-14:
Currently am lacking a webcam feed. All other values are working as expected including IP address capture. 

Testing on android w/ Chrome

kennethjiang, 2017-08-14:
Thanks for submitting this issue @modem7.

Can you let me know the email you logged in with, so that I can search
system logs to see what went wrong?

Also, was it a new installation or an upgrade from old version?

modem7, 2017-08-14:
Heya,

Upgrade from since before the server change.

Modem7 [at] gmail.com

I've also re-registered as well since the server change.

Thanks!

kennethjiang, 2017-08-15:
Can you open your octoprint.log (if you have trouble locating it please let me know) and look for possible error messages. If you find error messages there, please send them to me so that I can take a look at them. You can attach the whole log file if you don't mind.

I also sent you email. Please respond to my email in case you don't want to post logs publicly here in github. @modem7

kennethjiang, 2017-08-16:
Just found out that you opened an issue before and apparently you were getting webcam feed until some version upgrade. Definitely let me work with you to figure out what version broke your setup.

modem7, 2017-08-16:
Hi Kenneth!

What I'll do before we have a chat, is uninstall the current implementation
and reinstall. Sods law I broke something tinkering around!

I'm just doing a 10 hour print now, so tomorrow I'll do this and let you
know the results!

kennethjiang, 2017-08-16:
Sounds good!

Sabreur, 2017-08-17:
Having the same issue - everything works, but it gets stuck on "Waiting for camera feed..."

I've attached my latest log file.

[octoprint.txt](https://github.com/kennethjiang/OctoPrint-Anywhere/files/1230279/octoprint.txt)

modem7, 2017-08-17:
Looks like a reinstall and re-register didn't do the job in this case unfortunately.

kennethjiang, 2017-08-17:
@Sabreur I found the problem from the log file you posted. Thank you so much!

I just fixed it and released version 0.5.1. Please let me know if you still experience problems after you upgrade the plugin.

In case OctoPrint failed to push the upgrade to you, you can use this link to force upgrade: https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip

modem7, 2017-08-17:
Damn, unfortunately still no go here.
[octoprint.txt](https://github.com/kennethjiang/OctoPrint-Anywhere/files/1232015/octoprint.txt)

kennethjiang, 2017-08-17:
Unfortunately your problem seems to be caused by a different and trickier issue. @modem7.  The log you uploaded doesn't have the errors in @Sabreur's log file. So I'm pretty sure 0.5.1 won't fix it for you.

I suspect that your issue is on the browser side, which log file can't capture. The easiest way to debug it is via a screen-sharing session.

Sabreur, 2017-08-18:
@kennethjiang Just wanted to say that 0.5.1 fixed my issue, thank you very much!

kennethjiang, 2017-09-19:
@modem7 I'm closing this issue now assuming that you no longer have problems. Feel free to re-open it and provide more info if it's not the case.

closed 2017-09-19 by kennethjiang