kennethjiang/OctoPrint-Anywhere #33: Temp and time

strongbow1, 2017-11-05:
Is it possible that if you share your print that the people with the link can see Temperature and also the time till the print is completed?

kennethjiang, 2017-11-05:
It is not hard. The reason why I didn't do it was because I was not sure if some users will be concerned about sharing too much. :)

I can see why sharing time to complete is useful. But can you elaborate why you'd like to share temps as well @strongbow1 .

strongbow1, 2017-11-05:
To be honest I so have no idea why I mentioned Temp. I think time would be enough. How long its been printing and how much time is left and if possible how many % is completed. I think those are enough so the person you share with has this info also. Thank you for your reply

Mit Freundlichen Grüsse
With Best Regards
Sandro Strimer

kennethjiang, 2019-06-27:
Sorry for not getting around to this feature since there haven't been many users asking for it. I'm closing this for now and you can re-open it if you feel strongly about this feature.

closed 2019-06-27 by kennethjiang