kennethjiang/OctoPrint-Anywhere #36: Feature: gCode viewer and file browser?

calonmerc, 2017-11-08:
Wondering if there's anything on your roadmap for implementing a gCode viewer (similar to that in Octoprint currently, uses a socket stream to pass data continuously)?

Or any thoughts on implementing a file browser per-printer that would allow starting the print remotely?

kennethjiang, 2017-12-04:
@calonmerc sorry for the delay. I was just back from an extended trip this week.

I guess your request to have gcode viewer and to remotely start print job are related. So I'm trying to address them both here.

I have not considered remote job starting as a feature on the roadmap because I think, more often than not, a user will have to do some work (taking previous print off the plate, leveling the bed, making sure the filament is what you want, etc) on the printer before kicking off the job.

Let me know if you do freqently need to start print job remotely.

calonmerc, 2017-12-07:
I personally do, as I tend to get jobs for printing when I've left the house, but I always leave my printer clean and ready to go. Having to change filament would change that, as well, but generally I print in the same filament for most things. Otherwise, remote gcode would be more appreciated!

kennethjiang, 2017-12-08:
Ok I see the reasons why remote starting will be very helpful to you.

Both remote starting and gcode viewer can be quite complex to implement. I will put them on the roadmap and prioritize them against other features.

beeb, 2019-02-28:
Just commenting to propose another scenario where being able to start a job from Octoprint Anywhere could be useful. Imagine a raspberry pi running Octoprint is connected to the network by a cat5 cable. The workstation with access to the same network is in a different room or even different floor. The operator goes to the printer to load filament or clean build plate, and then wants to start printing from the Pi's SD. Since there is no wifi for the phone to connect to, access to OctoPrint can only be done in another room or floor. It would be super convenient if one could start the print from a phone connected to mobile internet.

kennethjiang, 2019-02-28:
@beeb thank you for providing this interesting use case. Will definitely keep it in mind.

kennethjiang, 2019-06-27:
@calonmerc  @beeb GCode upload and remote print start have been added to OctoPrint Anywhere 1.0. If GCode viewer, which takes quite a bit of work to do, is highly desired, please open another issue for it.

closed 2019-06-27 by kennethjiang