kennethjiang/OctoPrint-Anywhere #37: Plugin conflict?

knightofkairell, 2017-11-20:
I was installing multiply plugins and after I installed yours it broke access to the anything on the browser. it displays the images but when you click anything, nothing reacts. I saw you have another open issue with file manager, could this be related? Up until I installed your plugin and reset I still had access to everything. Now I have no idea how to go back and uninstall your plugin so I can regain use of the browser. Is there a SSH command line that can be used to uninstall it? I really don't want to have to re-image my SD card again. Thanks.

kennethjiang, 2017-11-21:
I'm currently on vacation with limited bandwidth. Will take a look when I'm back next week.

kennethjiang, 2017-12-04:
Hi @knightofkairell I'm back from my vacation. 

You should be able to `ssh` to your Raspberry Pi and run `/usr/bin/python -m pip uninstall --yes OctoPrint-Anywhere` from the command line to uninstall this plugin. 

Can you also do me a favor to let me know the OS/browser you are running on so that I can come up with a performanent solution for this issue?

kennethjiang, 2017-12-06:
Hi @knightofkairell I just released version 0.6.0 which uses Javascript IIFE to minimize potential conflicts with other plugins. Can you upgrade/re-install the plugin and let me know if you still experience the same problem?

Thanks!

closed 2017-12-06 by kennethjiang