kennethjiang/OctoPrint-Anywhere #39: Logs gets swamped with "backoff - ERROR - Backing off __probe_auth_token__(...) for $xs (HTTPError: 401 Client Error: Unauthorized)"

foosel, 2017-11-21:
I just received an `octoprint.log` file from a user (to look into an unrelated problem) and noticed that the `backoff` dependency of this plugin is swamping the log with an endless stream of lines like this one:

```
2017-11-19 12:14:35,371 - backoff - ERROR - Backing off __probe_auth_token__(...) for 0.4s (HTTPError: 401 Client Error: Unauthorized)
```

basically every couple of seconds. That makes the log pretty much unusable for debugging anything. I figure this is related to some kind of authorization error with the server component, but could it maybe get toned down a bit to not produce MBs of pretty much unhelpful log lines that make everything else drown in noise?

kennethjiang, 2017-11-21:
Hi @foosel  - this error log only happens when there is a problem with that particular user/installation. It is very repetitive but is also helpful for troubleshooting.

When the plugin works correctly, which is the case for 99+% of the users, the plugin put nothing to the log file.

Of course I'd like to keep the log size down even when there is an error. But it's not clear to me how that can be achieved. This is error came from a package called `backoff`. I already set the re-try policy to be exponential. If there is a way to further reduce the size of log in error case, please let me know and I'd love to implement it.

foosel, 2017-11-22:
This exponential retry appears to not work then. The log file in question literally contains error lines every 1 to 5 seconds over the course of hours. It also doesn't stop. If stuff is broken beyond repair I'd expect it to not keep trying and spam the log for all eternity. I understand that this is a third party library you are using here and that you have limited control over. What I'm saying is that something is not behaving correctly here in case of broken scenarios. I can provide the log in question if that helps you. The nuclear option to solve this would be to force a log level of fatal for the logger in question (once this kind of scenario is encountered), instead of error. I understand that this is not ideal, hence the idea to limit it when things look like they can't possibly work.

What I'm afraid of as things are right now is wear and tear of SD cards and also filled up file systems when the issue DOES in fact arise. I know it usually doesn't, but when it does it causes a problem.

foosel, 2017-11-22:
Here's the log: https://gist.github.com/foosel/faacac400bb3314f785704030aad0e48. Warning: It's big.

kennethjiang, 2017-11-22:
Every 1-5 seconds doesn't sound right. I will take a look at the log and figure out what went wrong.

kennethjiang, 2017-12-06:
Should be fixed in version 0.6.0.

closed 2017-12-06 by kennethjiang