kennethjiang/OctoPrint-Anywhere #40: Constant upload is killing household bandwidth

starxraider, 2017-12-06:
So, I installed this on friends RaspPi Octopi setup, and yesterday he showed me that his internet upload is almost the same as his download.

Compared to all the devices which have a high download, the Pi specifically had an extremely high upload. 

The limit isn't the problem, but it's slowing down his other devices that require upload/download. 

Is there any solution to this? Or can you make any recommendation's other than disabling the plugin when not in use?

kennethjiang, 2017-12-07:
Thanks @starxraider for submitting this issue.

Unfortunately I don't have a good solution to this problem. One workaround is to lower the resolution of the webcam so that less data needs to be transmitted. I set the resolution of my webcam to 1280x720, and the plugin is uploading at the rate lower than 3Mbps. Even though my cable connection only has 10Mbps uplink rate limit, it's not been a problem for me since I don't have a lot of other apps/devices fighting for upload bandwidth. But in case upload bandwidth became a problem to me, I'd probably just change the resolution to 640x360 and the bandwidth usage will be only 1/4 of what it is today.

Do you know what's the uplink rate limit in your friend's house, and the resolution of his webcam?

LilBub, 2018-02-24:
I have a 10 mb upload via my home cable company and with 4 printers using this, my upload speed is getting crushed.  I have to down the Pi's when not in use

kennethjiang, 2018-02-24:
@LilBub Yes I can see how 4 printers will jam all the uplink bandwidth. Will that be an option to reduce the resolution to 640x360?

There are some options to reduce the bandwidth requirement, but none of them can be done without complicating the system by a lot. So I'm hoping that a lower resolution will be a good-enough workaround...

LilBub, 2018-02-24:
I'd really hate to lower the resolution.  Perhaps an option to only broadcast when a print is running?

kennethjiang, 2018-02-25:
That'll be definitely part of the solution. However, I don't think it's a good idea to not show webcam streaming at all when the printer is not printing, as it'll make it as if the printer is offline.

So I guess the final solution should also stream webcam when it's not printing, but at a lower rate. Now the side effect is big gaps between webcam frames and hence long latency for webcam stream to show up on the dashboard upon page load...

LilBub, 2018-02-25:
My suggestion would be to allow there to be an option (Yes/No) on whether one wants streaming on all the time or only when a print is being made.  That would give those of us with lower bandwidth upstream to use the plugin and down it when not needed.  Users like options :)

hakan42, 2018-02-26:
Yes, a button for "Start streaming now, and keep streaming for the next 5 minutes" would be nice. There is already the button for start/cancel print, so communicating with the octopi seems to be already possible :)

GrueMaster, 2018-03-03:
Not sure how the remote portion works, but would it be possible to just upload a static image at a long interval and then a simple heartbeat every few minutes?  The upload bandwidth would be reduced dramatically this way.

Of course, switch to live when actually printing.

kennethjiang, 2018-03-03:
@GrueMaster Thanks for the suggestion. That sounds like solution that won't cost me an arm and a leg to build. Will work on it when I have a chance.

kennethjiang, 2018-03-15:
@starxraider @GrueMaster @hakan42 @LilBub I released version 0.6.1 that will slow down the streaming to ~5% of original bandwidth when not printing. Please upgrade and let me know if it solves or alleviate the problem.

closed 2018-03-15 by kennethjiang

kennethjiang, 2018-07-13:
@starxraider @GrueMaster @hakan42 @LilBub I released version 0.7.0 that further slows down the upload bandwidth when the webcam feed is not being watched. Please upgrade.

LilBub, 2018-07-14:
Thanks...I just installed this last night and suspect I have it...didn't notice the whole house lagging from two printers running this time!

kennethjiang, 2018-07-14:
I am glad to hear that 😁