kennethjiang/OctoPrint-Anywhere #41: Feature: Public viewing

JD2005L, 2017-12-11:
Wondering if you're considering implementing a hard link (or even time limited expiring links) that could be given to people for unauthenticated viewing with no admin access.  I'd like to provide viewing links to customers so they can watch their part be manufactured and this would be perfect.  Either with a static hard link, or preferably have the ability to create custom links that expire after a defined time period (1 day, 3 days, 5 days, etc...).  Thanks.

kennethjiang, 2017-12-12:
You can currently share the webcam feed with a link. Let me know if that feature is not very discoverable to you @JD2005L . Also if you need to share other things in addition to just webcam feed, please let me know too!

JD2005L, 2017-12-12:
Sorry about that, I see it now.  This is great, thanks.  The only extra
enhancement I would find useful would be if I could create multiple links
and set usage expiration dates for them so that each link was only active
for a specified period of time.

Thanks for your work on this though, very useful!

kennethjiang, 2017-12-12:
Other users have mentioned the similar need to have multiple links expiring at different times. However it's a pretty advanced feature and much trickier to implement in a right way. I will open another issue to keep track of it but it'll take a while before I can get around to it.

closed 2017-12-12 by kennethjiang

JD2005L, 2017-12-12:
No problem, appreciate your work!  I just think that feature is what stands
between this product being useful to an individual vs being able to be
leveraged to support a printing business.  I'd be hesitant to provide a
public link to customers and not be able to ensure that they don't have
perpetual ability to consume my bandwidth.

Thanks, hope to see this feature in the future!

kennethjiang, 2017-12-13:
I'll love to build these features at much faster pace and make the service a lot more robust if I can figure out a way to make it a self-sustainable business. Can you send an email to me at `kenneth dot jiang at gmail dot com` so that we have a google hangout chat about it @JD2005L ?