kennethjiang/OctoPrint-Anywhere #42: Multiple links to share webcam feed with different expiration dates.

kennethjiang, 2017-12-12:


cnlson, 2018-04-03:
i was just going to log a request for this when i had someone that i sent the link to a while ago tell me it still worked.

kennethjiang, 2018-04-03:
Hi @cnlson  you should be able to remove the sharable link, as in the attached screenshot.
[image]

Let me know if the link is still active even after you turn it off.

cnlson, 2018-04-03:
Yes but I'd like to be able to share when I'm doing a project for someone and don't want to continue sharing months down the line. 

Maybe generate a new hash each time it is turned off and back on?

kennethjiang, 2018-04-03:
Every time you turn the sharing off and turn it on again it'll be a different link. Let me know if it's not the case for you @cnlson

cnlson, 2018-04-03:
you are correct. I don't know if i hadn't disabled it in a while or what. but thank you!