kennethjiang/OctoPrint-Anywhere #43: Waiting for Camera Feed

strongbow1, 2018-01-03:
After registering my camera I only get waiting for camera feed. But my Camera works fine within Octoprint

kennethjiang, 2018-01-04:
Sorry I was just back from vacation.

Do you know how you to get access to OctoPrint log files? If so, can you zip up those log files and post them here or send them to me at kenneth dot jiang at gmail dot com? @strongbow1

strongbow1, 2018-01-04:
Hi kenneth

Thank you for your reply. Sorry I do not know how to access the Octoprint log files. If you tell me how to get them I will get them for you.

Mit Freundlichen Grüsse
With Best Regards
Sandro Strimer

strongbow1, 2018-01-05:
> Assuming your OctoPrint runs in a Raspberry Pi, you need to ssh to the Raspberry Pi, go to folder /home/pi/.octoprint/logs. The OctoPrint log files are octoprint.log and other files such as octoprint.log.2017-xxx. Please zip up all these files and send them to me.

Hi Kenneth

I got it working again.

Just reinstalled octoprint on my raspi. That solved it for now

Thank you again for your help and fast response

Mit Freundlichen Grüsse
With Best Regards
Sandro Strimer

kennethjiang, 2018-01-05:
You are welcome! I'm glad that you figured this out.

closed 2018-01-05 by kennethjiang