kennethjiang/OctoPrint-Anywhere #44: No such app

Stellarspiff, 2018-01-20:
It stopped allowing me to view the Octoprint Anywhere page. Sends me to a Heroku page that says "There's nothing here, yet."

kennethjiang, 2018-01-21:
shoot there is an outage on the backend. working on that!

Stellarspiff, 2018-01-21:
Thanks for the update!

johnpettybone, 2018-01-21:
Same problem. Eta? :)

kennethjiang, 2018-01-21:
Sorry it's quite a nasty backend issue. Will take a while. working on it but can't provide an eta now...

kennethjiang, 2018-01-21:
@Stellarspiff @johnpettybone I restored the backend service. However it requires up to 24 hours to get the new DNS to propagate. Please let me know if you still can't access the Dashboard after 24 hours. Sorry for the trouble!

kennethjiang, 2018-01-21:
It looks like the backend is restored for most users around the world. Closing this issue.

closed 2018-01-21 by kennethjiang