kennethjiang/OctoPrint-Anywhere #46: Webcam feeds not showing up

cptprime, 2018-02-12:
i have 7 identical printers all running their own Pi/octopi setup, i thought i would be smart and have yawcam streaming a single webcam across the local network to all the octopis, it works well via octopi but on the dashboard it gets stuck on "Waiting for camera feed..."

kennethjiang, 2018-02-13:
Hi @cptprime thank you for submitting the issue.

Do you have 7 OctoPrints (7 Rasperry Pis), or only 1? 

I'm not familiar with yawcam. Can you provide a little more info on how the yawcam is connected to Raspberry Pi(s)?

cptprime, 2018-02-14:
Hi @kennethjiang i have indeed 7 raspberry Pis, yawcam just serves my webcam as a stream on the local network, it also lets me do timelapse from within octoprint.

i just point octoprint's webcam adress from 127.0.0.1 to what ever the adress of yawcam is

kennethjiang, 2018-02-15:
Ah I understand now.

I just checked and yawcam and it is a MJPEG server. Assuming yawcam can be displayed correctly in the `Control` tab in OctoPrint, I can't think of a reason why it wouldn't work with OctoPrint Anywhere. 

Do you install OctoPrint Anywhere on all OctoPis, or just one of them?

kennethjiang, 2018-07-30:
Hi @cptprime  I'm cleaning up old issues. Do you still experience this issue? If not I'd like to close it.

closed 2018-08-06 by kennethjiang