kennethjiang/OctoPrint-Anywhere #47: Full screen webcam?

Quintin9801, 2018-02-14:
A full screen webcam feature would be amazing, I’m currently just zooming in on the image but this enhancement would be nice!

kennethjiang, 2018-02-14:
Very good suggestion! Will get to it at some point!

Quintin9801, 2018-02-14:
Awesome! Thank you so much

JD2005L, 2018-02-20:
Came here to ask for the same, would be very useful.  Double click video for fullscreen, for example, similar functionality to Octoprint-Fullscreen plugin: https://github.com/BillyBlaze/OctoPrint-FullScreen

kennethjiang, 2018-02-22:
Thx @JD2005L  I will take a look at that plugin to see how it's implemented

kennethjiang, 2018-04-12:
@JD2005L @Quintin9801 full screen turns out to be non-trivial to do. But now the image is zoomable (https://github.com/kennethjiang/OctoPrint-Anywhere/issues/54). I hope this at least solve part of your problem.

JD2005L, 2018-04-12:
Yep works for me.  Thanks!

kennethjiang, 2019-06-27:
@JD2005L The new 25fps video streaming will go full screen on a tap. Closing this issue.

closed 2019-06-27 by kennethjiang