kennethjiang/OctoPrint-Anywhere #48: A way to arrange cam windows

mylife4aiurr, 2018-02-23:
I love your pluggin. Its very useful when Im away from printers. Its piece of mind. Ive cancelled prints remotely that failed or lost adhesion. But also to know a print so doing fine. Great to show co-workers.

My request is for a way to choose and view multiple printers. Because not all printers are always printing however. 

So currently if printer A and C are printing, B and D is shown idle. Tomorrow D and B maybe printing but A and C are showing @ https://www.getanywhere.io/app/devices, with idle screens. There is no way to size the web screens smaller. So currently I can only view 2 screens without scrolling. In the four printer scenario, A's printing, B idle, C's printing, D idle. But the standard 800 x 600 or even 1920 x 1080 web browser page will show 2 printers max without scrolling, and 2 could be idle:

Web page:
A printing B idle(waste)
must scroll
C printing D idle

wish i could choose active printers:
A printing C printing
B idle D idle

or 3 printers printing
Choose active printers - scale smaller to view A, C, D printing same page.

--------------------------
Suggestion 1st page -
Combine all printer cams single scaled window, choose which to zoom/expand. (similar to security office, where all cams can be viewed at once on 1 page)

then 

Need a way to group active printers, let user decide which printers are grouped. (so idle printer doesn't waste screen)

At home on local network, pi3 are connected by ethernet on a switch, not using the on board wifi of pi3 is more reliable for me, and I can stream locally @ 1024 x 720 25fps.
camera_raspi_options="-fps 25 -x 1280 -y 720 -ex auto -awb fluorescent -b 2000000 -ISO 200 -vf -hf"

Prusa Mk3 
Prusa MK2s
Taz 6
CR10s4
3ku Delta

kennethjiang, 2018-02-23:
Thank you @mylife4aiurr for a very constructive suggestion! Even though it sounds pretty difficult to implement, it definitely sheds lights on how I can further improve the user interface. I will think about it and come back with some options.

I don't fully understand your last part:

> At home on local network, pi3 are connected by ethernet on a switch, not using the on board wifi of pi3 is more reliable for me, and I can stream locally @ 1024 x 720 25fps.
camera_raspi_options="-fps 25 -x 1280 -y 720 -ex auto -awb fluorescent -b 2000000 -ISO 200 -vf -hf"

Is it something that is causing a different problem, or just the same problem but described differently?

mylife4aiurr, 2018-02-23:
i was just showing you the screen size i use with Octoprint. I think its why I only see two printers then have to scroll.

kennethjiang, 2018-02-24:
Great thanks!

mylife4aiurr, 2018-08-04:
I see you are now charging users for more then 1 instance. OK. I would consider paying but I still need this issue addressed. Have 5 printers.

I would need at a minimum to have actively printing printers presented 1st to user. So I dont have to scroll to find active printers. With printers doing nothing in between the ones printing.

Wish list:
Main screen with thumbnail sized windows of all printers. Click one that I'd like to view. 
I way to order the printers being viewed. Maybe I have 3 printers going. But I'd like to see the mk3 1st, taz6 2nd, delta 3rd. etc

kennethjiang, 2018-08-04:
Yes sorting the card is on my roadmap. I can definitely pull it in. The default sorting will be by status (active -> idle -> offline). You can also choose to sort by name. Does it sound like what you want?

By "clicking one to view", do you mean to view webcam feed in full screen? Or you mean screen with only 1 printer?

mylife4aiurr, 2018-08-06:
The sorting option is great as a minimum option.

This suggestion/option is for going beyond the minimum.... My wishlist.
If User has 3 or more printers.
Main screen is clickable icon/thumbnail view of all printers. 
If I want to full screen a printer, I can click on it's smaller pic/icon to get a full screen view (the screen u already have implemented) then I could stop print, raise head, etc - your current functionality.

kennethjiang, 2018-08-06:
@mylife4aiurr I just released a new version that includes sorting.

Is the reason why you want the thumbnail view because you want to see webcam feed from all printers at a glance? If so, I'm wondering if the thumbnail images will be too tiny for you to get the info you'd like to get anyway (such as if a printer is failing). Let me know if I misunderstood you.

mylife4aiurr, 2018-08-06:
Yes correct, my use case for multiple printer view is while away using a cell phone. If Octoprint doesn't support a good way to scale each small window for a particular phone or tablet screen then don't bother.

Thanks for implementing a sort function...

kennethjiang, 2018-08-06:
I'm not a big fan of "click to see detail / back button to thumbnail view" user interaction. I have been in many mobile-oriented projects and it's clear to me that card-based UI that requires mostly swipes  opposed to clicks is just much better user experience. I doubt that you would like that user experience even if I implemented the thumbnail/detail UI.

At some point I may implement the feature to let users toggle between thumbnail and card view, if I'm convinced that the webcam feeds on thumbnail view won't be too cramped to be useful.

Again I appreciate your constructive suggestions. I'll leave this issue open for now in case you want to provide additional info.

closed 2019-06-27 by kennethjiang