kennethjiang/OctoPrint-Anywhere #49: Printer active before the print started. Then offline.

rotaryfurball, 2018-03-04:
When i send off a print everything if fine with OctoPrint anywhere. But once the print starts the Octoprint-Anywhere says its offline. And it seems to break the whole Wifi in my house. Meaning there is Wifi signal but no data is going through, when the print is printing, so my Wifi is broken when the printer is printing.

kennethjiang, 2018-03-04:
Currently OctoPrint Anywhere plugin doesn't behave differently whether or not the printer is printing. Do it always happen that when your printer is printing, your wifi will be broken? Also by broken wifi, did you mean the whole wifi network is not usable so that no computers on the same wifi can connect to internet?

rotaryfurball, 2018-03-04:
yea. Once it starts printing the Octoprint Anywhere says the printer is offline. And yes all the computer s can't connect then now

kennethjiang, 2018-03-04:
Does it happen every time your printer starts printing? Does your wifi network come back every time your print is over? If both are true, you may want to uninstall OctoPrint Anywhere and see if the problem will go away.

kennethjiang, 2018-03-09:
@rotaryfurball One of my routers started to have the similar issue. I think it's probably because the plugin is sending too much data (please look at #40). I was able to solve the issue by switching OctoPrint to using a different router. If this is not an option for you, you probably have to uninstall this plugin, and try it again when #40 is fixed. Sorry!

kennethjiang, 2018-03-15:
@rotaryfurball I released version 0.6.1 that solved #40. Can you please upgrade and let me know if it also solves this issue?

rotaryfurball, 2018-03-19:
Hey sorry. The problem still occurs. I have attached pictures. The printer
is printing right now but it shows this( picture).
Sorry about the late reply, I was on vacation!

kennethjiang, 2018-03-20:
@rotaryfurball Can you uploaded the picture directly to github? Somehow it was lost so I can't see it.

rotaryfurball, 2018-03-20:
No worries. Will do when I get home

rotaryfurball, 2018-04-07:
[image]

kennethjiang, 2018-07-13:
@rotaryfurball I just released 0.7.0, which further reduces the upload bandwidth when webcam feed is not being watched. Please upgrade and let me know if you still experience the same issue.

kennethjiang, 2018-08-06:
Hi @rotaryfurball I assume that you either no longer use this plugin or no longer experience this problem. So I'm closing this issue for now. Feel free to re-open it.

closed 2018-08-06 by kennethjiang