kennethjiang/OctoPrint-Anywhere #51: clicking "Timelapses" in the getanywhere.io web app throws a JS error

gregsadetsky, 2018-03-21:
clicking "Timelapses" on getanywhere.io throws a JS error, and shows a blank screen. traceback below.

thanks for the great app/plugin!

```
app.27eac5.js:1 Uncaught (in promise) TypeError: Cannot read property 'id' of undefined
    at app.27eac5.js:1
(anonymous) @ app.27eac5.js:1
Promise.then (async)
value @ app.27eac5.js:1
value @ app.27eac5.js:1
CallbackQueue.notifyAll @ app.27eac5.js:1
close @ app.27eac5.js:1
closeAll @ app.27eac5.js:1
perform @ app.27eac5.js:1
batchedMountComponentIntoNode @ app.27eac5.js:1
perform @ app.27eac5.js:1
batchedUpdates @ app.27eac5.js:1
batchedUpdates @ app.27eac5.js:1
_renderNewRootComponent @ app.27eac5.js:1
_renderSubtreeIntoContainer @ app.27eac5.js:1
render @ app.27eac5.js:1
(anonymous) @ app.27eac5.js:1
__webpack_require__ @ app.27eac5.js:1
(anonymous) @ app.27eac5.js:1
__webpack_require__ @ app.27eac5.js:1
(anonymous) @ app.27eac5.js:1
(anonymous) @ app.27eac5.js:1
```

kennethjiang, 2018-03-21:
Hi @gregsadetsky Thank you so much for reporting this error! It's a silly bug in my code. I will find a chance to fix it and let you know as soon as I have done it.

kennethjiang, 2018-03-21:
@gregsadetsky Just fixed the bug. Just refresh the page and it should work. No upgrade needed. :)

closed 2018-03-21 by kennethjiang

gregsadetsky, 2018-03-21:
That's fantastic, just confirmed it to be working again. Thank you!