kennethjiang/OctoPrint-Anywhere #53: Is OctoPrint-Anywhere crashing my Octoprint service

DigChaos, 2018-04-05:
Love the plug-in, i'm really hoping this isn't what is causing my issue. I have 2 machines running this plugin in and both are mysteriously stopping after a few hours of printing intermittently. Here is a copy of the log https://pastebin.com/ePBFfTwh 

Any ideas?

kennethjiang, 2018-04-05:
From the logs it looks like somehow the connection from your OctoPi to the internet was shaky. The errors in logs were from every time when OctoPrint Anywhere tried but failed to connect to the server.

Although the connectivity issue doesn't seem to be caused by OctoPrint Anywhere plugin, I'll be happy to have a google hangout session to troubleshoot the problem. Feel free to email me at kenneth dot jiang at gmail dot com so that we can coordinate a time to chat.

kennethjiang, 2018-08-06:
@DigChaos It's pretty obvious now that the random crash issue is not caused by my plugin as it happens to other users without my plugin installed. This [OctoPrint issue](https://github.com/foosel/OctoPrint/issues/2647#issuecomment-391329662) has more details.

Also please note that I have improved the plugin a lot by reducing the bandwidth it requires. The new version barely sends any data when nobody is watching the webcam feed. Please upgrade if you haven't done so.

closed 2018-08-06 by kennethjiang