kennethjiang/OctoPrint-Anywhere #55: Login page is invisible

supersoulman, 2018-04-13:
The login page for getanywhere.io on iOS safari is invisible. 

https://i.imgur.com/tQj7tLZ.jpg

kennethjiang, 2018-04-17:
@supersoulman Sorry for the delay. Will take a look at it!

kennethjiang, 2018-04-25:
@supersoulman I tried it on Safari on an iPhone 8 but was able to see the login page. What iPhone are you using?

supersoulman, 2018-04-25:
I am using an iPhone 10. 
Here is a video of the issue. https://streamable.com/ijhl1

supersoulman, 2018-04-25:
Okay just discovered, while typing this comment, if I just change tabs in safari and come back the login boxes appears. 

https://streamable.com/irctq

kennethjiang, 2018-04-25:
Thank you so much @supersoulman for the gif. It made it clear to me what happened. 

I found a iphone x to try it and didn't have the same issue. So I'm not entirely sure why it's different on your iphone.

One thing that might help is: if you directly go to https://www.getanywhere.io/app/devices (I noticed that you went to https://www.getanywhere.io/pub first), you will directly go to the dashboard until your login session expires in 30 days. Hopefully this will make it less painful for you.

closed 2019-06-27 by kennethjiang