kennethjiang/OctoPrint-Anywhere #56: OctoPrint-Anywhere installs fine but isn't seen as a plugin

enotsone, 2018-04-22:
Using OctoPrint 1.3.8 (OctoPi 0.14) and using the plugin manager to install OctoPrint-Anywhere (0.6.1) the install goes fine with no errors.  When it is done, a dialog comes up saying 'Installed: unknown' and when the reboot of the OctoPrint service is done, the OctoPrint-Anywhere doesn't show up under the tool menu.  When I go to install it again, it sees that it was already installed and does a 'force' install, yet it still shows as 'Installed: unknown' and when rebooted, nothing shows up.

During the first install, it stated that my pip version was a bit old, so I updated it but still no joy.
Matt
[image]
[image]

kennethjiang, 2018-04-23:
It looks like OctoPrint 1.3.8 somehow broke the compatibility. Will working on fixing it asap.

ironjoe13, 2018-08-21:
I have this same problem with 0.15. octopi and 1.3.8 Octoprint

kennethjiang, 2018-08-22:
Hi @ironjoe13 Can you post your `octoprint.log` file here? Refer to [this document](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299) for how to do download the log file.

kennethjiang, 2018-08-22:
Awesome thanks!

ironjoe13, 2018-08-22:
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2311441/octoprint.log)

kennethjiang, 2018-08-22:
From your `octoprint.log`, the installation of OctoPrint Anywhere succeeded at 2018-08-21 17:24:27 (UTC), and the status of the plugin changed from `disabled` to `active` after OctoPrint restarted.

Can you see OctoPrint Anywhere config page as below?
[image]

ironjoe13, 2018-08-22:
It does not show up. I just tried to install it again after it was uninstalled yesterday and got this with the Unknown plugin notice:
[image]

ironjoe13, 2018-08-22:
[plugin_pluginmanager_console.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2311782/plugin_pluginmanager_console.log)

kennethjiang, 2018-08-23:
The failed installation happened at  2018-08-22 19:46:20 UTC. The succeeded installation in your previous `octoprint.log` happened at 2018-08-21 17:24:27 UTC. Were these 2 installations on the same OctoPrint (1.3.8 I assume)?

What did you see after the succeeded (at least according to `octoprint.log`) installation at  2018-08-21 17:24:27 UTC? Did you still not see the OctoPrint Anywhere config dialog? Was it why you had to uninstall it and install it again at 2018-08-22 19:46:20 UTC ?

ironjoe13, 2018-08-23:
Octoprint Anywhere was visible and working on the 1.3.9 install onto separate Raspbian system. Yes I did uninstall and reinstall, no change.
The "successful" install at 17:24 was followed by : 
2018-08-21 17:31:58,020 - tornado.access - WARNING - 403 POST /api/plugin/anywhere (127.0.0.1) 14.19ms
2018-08-21 17:45:49,075 - tornado.access - WARNING - 403 POST /api/plugin/anywhere (127.0.0.1) 14.73ms
It did not show up after that installation either, after saying it was "unknown plugin"

kennethjiang, 2018-08-23:
Thank you so much for catching those 403 errors.

There are some existing OctoPrint issues that may be similar to this. In https://github.com/foosel/OctoPrint/issues/2184 it was suggested it could be caused by a permission issue in `/tmp/` folder.

https://github.com/foosel/OctoPrint/issues/1881 suggests that this issue should be fixed in 1.3.9

If none of these solutions helped, feel free to log a bug for OctoPrint at https://github.com/foosel/OctoPrint/issues/new . I will be happy to provide whatever info @foosel needs to get to the bottom of it.

The second error (-3 exit code by decompressing) seems like a separate issue. It also seems like a temporary error. Have you seen this error dialog more than once?

ironjoe13, 2018-08-23:
I have gotten the temporary error before, I haven't figured out why as of yet.... I believe it has something to do with when I am on YouTube on one browser page, and Octoprint on another page(separate displays, displaying both pages at once)
I will update to 1.3.9 and let you know what happens

kennethjiang, 2018-11-05:
@ironjoe13 are you still experiencing this error?

onemadpunk, 2018-11-07:
Hi there i am having trouble installing your plugin as well. andy help would be appreciated as it seems a great addition to octoprint.

Installing plugin "OctoPrint Anywhere" from https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip install https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip --no-cache-dir
Collecting https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip
Downloading https://github.com/kennethjiang/OctoPrint-Anywhere/archive/master.zip
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

[plugin_pluginmanager_console.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2557341/plugin_pluginmanager_console.log)

Thanks in advance for your help
octoprint Version 1.3.9
octopi Version 0.15.1

kennethjiang, 2018-11-07:
Hi @onemadpunk , the log file shows that it failed to download the plugin zip file from github. The most likely cause was that your Pi couldn't connect to internet at that time. Please make sure your Pi can connect to internet and try it again. Please note that it's possible that your computer can connect to internet but your Pi can't. So the best way to test its internet connection is to ssh to the Pi and test from there.

onemadpunk, 2018-11-07:
Hi there 

Updated the PIP to 18 not sure what that is but it was the error on screen
/home/pi/oprint/bin/python -m pip install --upgrade pip

it then installed the plugin.

in octoprint settings i only see
PLUGINS
Action Command Prompt
Bed Leveling Plugin
CuraEngine (<= 15.04)

if i go to plugin manager i can see your plugin installed and enabled
OctoPrint Anywhere (0.9.0)
Control your OctoPrint from anywhere, using your smart phone
 Homepage  AGPLv3  Kenneth Jiang  

however I can see nowhere to follow your setup after install. it did ask me to restart which i have done.

thanks for your quick reply it is nice to see you really do support your work, its much appreciated

onemadpunk, 2018-11-07:
Hi there I have the issue that it is now showing as installed in plugin manager but not seen anywhere else. I cant access the plug in. 

It does not show on the plug in list. and a did not get a set up wizard. I have tried re installing but still the same issue.

Any idea what I am doing wrong ?

Thanks again

kennethjiang, 2018-11-07:
Oh sorry I thought your issue was about being unable to install. Re-opening this issue.
Did you see the OctoPrint Anywhere settings page, as in the screenshot attached?
[image]

onemadpunk, 2018-11-07:
[image]

Hi there I hope this helps show the issue. 
Thanks
Steven

kennethjiang, 2018-11-07:
Ah I got it now. Thank you so much @onemadpunk for the screenshot!
Can you send your `octoprint.log` file to k@getanywhere.io? I'm hoping that file can provide some insights.

onemadpunk, 2018-11-08:
Hi there I have emailed the file as requested, thanks again for the help.

kennethjiang, 2018-11-09:
Thank you so much @onemadpunk for letting me debug it on your pi and figured this out.

@ironjoe13 @enotsone  if you are still having this issue, you can try the solution that worked for @onemadpunk 
- SSH to your Pi.
- run `/home/pi/oprint/bin/pip uninstall raven`
- Uninstall and re-install OctoPrint Anywhere plugin.

closed 2018-11-09 by kennethjiang