kennethjiang/OctoPrint-Anywhere #57: Cancel not working

MoruRising, 2018-05-15:
Have been doing a few tests, and i can get octoprint to cancel from the local page. But when using the plugin it does not cancel the print. The temperature continues to stay at target but the positioning stops.

kennethjiang, 2018-05-15:
What OctoPrint version are you using @MoruRising ? I tested cancellation function on my OctoPrint (1.3.8) just now and it worked...

Also if you have not upgraded to the latest version of OctoPrint Anywhere plugin, please do so.

kennethjiang, 2018-05-15:
From the log it appears that when you canceled the job using the plugin, there was a serial error. That serial error could have happened to be there, but more likely it was because OctoPrint Anywhere plugin sent the cancel order in a slightly different way than OctoPrint itself.

Can you find the `serial.log` with the timestamp that is just before the last one? The `serial.log` in your zip only has the log after OctoPrint rebooted and missed the time when cancelation happened. Thanks!

MoruRising, 2018-05-15:
Sure thing, here you go!
[serialLogs.zip](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2006497/serialLogs.zip)

kennethjiang, 2018-05-15:
It looks like canceling through the plugin will cause OctoPrint to send different gcode from directly cancelling it in OctoPrint. Will investigate more and get back to you on it.

kennethjiang, 2018-05-16:
Hi @MoruRising I guess your printer's firmware is different from mine so OctoPrint sends slightly different Gcode to cancel a job. After examining `serial.log.2018-05-15_17-09-56` you sent, here is what I found out:

When you cancelled the job directly in OctoPrint, the Gcode sequence is:
```
 40 2018-05-15 17:08:34,670 - Changing monitoring state from "Printing" to "Cancelling"
 41 2018-05-15 17:08:34,671 - Send: N13 M18**2
 42 2018-05-15 17:08:34,677 - Send: N14 M109 S215**113
 43 2018-05-15 17:08:34,695 - Send: N15 G4 S15**68
 44 2018-05-15 17:08:34,703 - Send: N16 M106 S255**120
 45 2018-05-15 17:08:34,719 - Send: N17 M17**9
 46 2018-05-15 17:08:34,721 - Recv: T:96.9927
 47 2018-05-15 17:08:34,794 - Send: N18 G0 Z1.310794 F48**61
 48 2018-05-15 17:08:34,824 - Send: N19 G92 E0**85
 49 2018-05-15 17:08:34,831 - Send: N20 G0 Z1.310794 E10 F360**107
 50 2018-05-15 17:08:34,833 - Send: N21 G4 S3**116
 51 2018-05-15 17:08:34,848 - Send: N22 G0 X0.240000 Y2.100000 F1500.000000**123
 52 2018-05-15 17:08:34,851 - Send: N23 G0 X5.640000 Y6.950000 Z-1.119248 F400**35
 53 2018-05-15 17:08:34,863 - Send: N24 G92 E0**91
 54 2018-05-15 17:08:34,876 - Send: N25 G0 X54.910000 Y34.309000 Z0.029264 F1740**58
 55 2018-05-15 17:08:34,883 - Send: N26 M84*43
 56 2018-05-15 17:08:34,886 - Send: N27 M104 T0 S0*20
 57 2018-05-15 17:08:34,896 - Send: N28 M106 S0*93
 58 2018-05-15 17:08:34,907 - Changing monitoring state from "Cancelling" to "Operational"
````

When you cancelled it using the plugin, here is the Gcode:
```
125 2018-05-15 17:09:18,275 - Changing monitoring state from "Printing" to "Cancelling"
126 2018-05-15 17:09:18,350 - Send: N12 G0 X0.000000 Y1.500000 Z4.110694 F1800**49
127 2018-05-15 17:09:33,326 - Recv: ok 11
128 2018-05-15 17:09:33,361 - Send: N13 M18**2
129 2018-05-15 17:09:33,371 - Recv: T:154.2919
130 2018-05-15 17:09:33,377 - Recv: ok 12
131 2018-05-15 17:09:33,388 - Send: N14 M109 S215**113
132 2018-05-15 17:09:33,735 - Recv: T:156.8356
133 2018-05-15 17:09:34,738 - Recv: T:158.8327
134 2018-05-15 17:09:35,745 - Recv: T:162.0912
135 2018-05-15 17:09:36,383 - Recv: ok 13
136 2018-05-15 17:09:36,418 - Send: N15 G4 S15**68
137 2018-05-15 17:09:36,776 - Recv: T:145.6416
138 2018-05-15 17:09:37,781 - Recv: T:149.4494
139 2018-05-15 17:09:38,783 - Recv: T:151.5726
140 2018-05-15 17:09:39,789 - Recv: T:152.8564
141 2018-05-15 17:09:41,796 - Recv: T:155.9884
142 2018-05-15 17:09:42,799 - Recv: T:156.7360
143 2018-05-15 17:09:43,803 - Recv: T:159.4418
144 2018-05-15 17:09:44,806 - Recv: T:160.7533
145 2018-05-15 17:09:45,811 - Recv: T:161.3171
146 2018-05-15 17:09:46,814 - Recv: T:162.5440
147 2018-05-15 17:09:47,830 - Recv: T:165.3821
148 2018-05-15 17:09:48,821 - Recv: T:166.7157
149 2018-05-15 17:09:49,825 - Recv: T:168.9010
150 2018-05-15 17:09:50,830 - Recv: T:170.1585
151 2018-05-15 17:09:51,833 - Recv: T:171.4232
152 2018-05-15 17:09:53,840 - Recv: T:174.9066
153 2018-05-15 17:09:54,844 - Recv: T:177.0418
154 2018-05-15 17:09:55,847 - Recv: T:175.9476
155 2018-05-15 17:09:56,222 - Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports re    adiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2417
156 2018-05-15 17:09:56,225 - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
157 2018-05-15 17:09:56,242 - Changing monitoring state from "Cancelling" to "Offline (Error: SerialException: 'device reports readiness to read but re    turned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2417)"
158 2018-05-15 17:09:56,244 - Connection closed, closing down monitor
159 2018-05-15 17:09:56,262 - Closing down send loop
```

I'm not an expert on 3D printer gcode, but it appears to me that somehow when the print job is cancelled using the plugin, OctoPrint stopped after it sent `N15 G4 S15**68`.

In the plugin, I am using the documented OctoPrint API (http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.cancel_print) to cancel the job. It's possible that OctoPrint has subtle different behavior depending on if this api is called by OctoPrint itself or by a plugin. But since my printer doesn't have this problem, the best way to find it out is to debug it in your OctoPrint.

Let me know if there is a way for me to directly email/call you so that we can debug it together in your OctoPrint.

MoruRising, 2018-05-16:
Thanks for looking into that! And sure thing, feel free to email me at dreamevolved@gmail.com.

kennethjiang, 2018-07-30:
Closing this ticket since this seems to be a combination of OctoPrint and your old M3D printer. @MoruRising Feel free to reopen it if this is still an issue for you.

closed 2018-07-30 by kennethjiang