kennethjiang/OctoPrint-Anywhere #61: Camera feed freezing

midashand33, 2018-06-20:
Installed OctoPrint Anywhere yesterday and it is very handy. Already has allowed me to stop a print that came loose from the bed before it wasted a bunch of filament.

However, the page seems to be freezing and/or dropping the video feed somewhere between 10 and 60 seconds after the page loads. Sometimes the video grays out slightly, sometimes it just freezes, sometimes it disappears entirely.

I was using Chrome, and haven't really had a chance to test other browsers very well. Is this a common issue? I skimmed over the current open issues and I didn't really see anything directly relating to this.

Apologies if I'm missing something simple, its been a long day.

midashand33, 2018-06-20:
Additional information: I am running on a Raspberry Pi 3 model B and streaming over WiFi. No other devices are on the network at the time. (I put my desktop to sleep and everything is turned off or taken with me when I leave for work.)

kennethjiang, 2018-06-20:
Thank you @midashand33 for submitting this issue.

OA streams webcam feed at ~3 frames/sec when it's printing, and slows down the rate to about 1 frame/10s when it's not printing to save the bandwidth. So if the feed freezes for about 10s when it's not printing, it's quite normal. However if it freezes when it's printing, please let me know so that I can take a close look.

The video grays out when the dashboard page tries to talk to OA plugin but have not heard back from it, to indicate *I'm not sure if the printer is online or not*. Again if it stays greyed out for longer than a few seconds while you are sure the printer is online (connected to internet), please let me know and I'll look at it.

midashand33, 2018-06-20:
This is while it is printing. I had an 11+ hour print that I was monitoring today at work. Refreshing the page fixes it temporarily, but it freezes again after a minute or so.

kennethjiang, 2018-06-20:
It does sound like you are running into connectivity issues. 

Can you find the`octoprint.log` that covers the period when the video froze and post it here? You can check https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299 for how to do it.

kennethjiang, 2018-06-20:
Hmmm it sounds like an issue if it disappears and reloads often. Does it do that only when you switch back to OA dashboard page, or even when you stay on the page?

midashand33, 2018-06-21:
I just have the dashboard page up on a second monitor. The drop and reload happens every hour...maybe two?

kennethjiang, 2018-06-22:
So it does sounds like a bug. Hopefully it's just annoying not really impacting the usefulness. I'll try to reproduce it and tackle it when I get a chance.

kennethjiang, 2018-07-16:
@midashand33 Please upgrade to version 0.7.0 if you haven't got a chance. There was a significant change in that version on how webcam feed is delivered that will not only result in lower bandwidth consumption, but hopefully also in a smoother feed. Let me know if it's the case for you.

kennethjiang, 2018-08-06:
@midashand33 I'm closing this issue since OA has improved a lot and this issue is most likely gone with all the bug fixes I put in. However feel free to re-open it if you still experience it.

closed 2018-08-06 by kennethjiang