kennethjiang/OctoPrint-Anywhere #62: Feature request : fluid video option

rvermeiren, 2018-07-01:
Hi, I have read you limit the frame rate of the video for bandwidth saving. I think it's a great idea but it could be nice to add an option to temporary increase the frame rate and have a good real-time view. Also, I have received your mail about the future of this project, and as a newly born dev (just finished my master degree), I could potentially give you some help to improve Anywhere ;-)  I will first start by reading how it's work. 

Regards, 

Rémy

kennethjiang, 2018-07-02:
I'll love to get your help! Feel free to send me email at k@getanywhere.io to further coordinate with me.

If/when you dig a bit deeper in the code, you can see that currently OA is sending about 3 frames/s when the printer is printing. And it slows down to  about 1 frame/10s when the printer is idle. This can be changed to something like "if user is watching it, send it at higher rate. Otherwise, don't send anything". If you are interested in doing this part, let me know and I'll tell you how to set up your dev environment so that you can start coding.

kennethjiang, 2018-07-13:
@rvermeiren I just released 0.7.0. The new version will stream webcam feed at full speed only when it's being watched, otherwise it'll steam at a much lower rate (~ 1 frame/ 20s).

closed 2018-07-13 by kennethjiang