kennethjiang/OctoPrint-Anywhere #63: Fee for Printers

pb777, 2018-07-21:
I got a message, that I need to pay 5$ for every printer beyond the first one! "You have more than 1 regisstered 3D printers. In the next couple weeks you will start to be charged by $5/printer/month for all printers beyond the 1st one"

1. I have only one printer registered
2. I checked the whole Page: Where is the hint that this service is chargeable?

Thanks!

kennethjiang, 2018-07-22:
Hi @pb777 Thank you for submitting this issue.

The service is free for now. But I'll have to start a non-free service tier soon to cover the cost of cloud service. I'll have the needed info on the website when I'm ready to do so.

What's the email address you used to register with OctoPrint Anywhere. I'll take a look to see why you had only 1 printer but still received the payment warning.

kennethjiang, 2018-07-22:
I found your account and yes you only have 1 registered printer. What you experienced was probably caused by a bug that I fixed about 8 hours ago. When you log in again if you still get this message, please let me know.

kennethjiang, 2018-07-30:
@pb777 Closing this issue now since I believe the issue is fixed.

closed 2018-07-30 by kennethjiang