kennethjiang/OctoPrint-Anywhere #64: Is there a way to shut off an Octopi server through this interface?

bwaslo, 2018-07-21:
I keep reading how the Raspi should be shut down by command before yanking power.  But could there be a way to do that from OctoPrint-Anywhere (or is there already?).

bwaslo, 2018-07-22:
> Hi @bwaslo <https://github.com/bwaslo> thank you for submitting a
> suggestion!
>
> I'm pretty sure this is doable when some conditions are met (for instance,
> OctoPrint is running under a user that has sudo privilege, which is true
> in most cases).
>
> However, this is also a "tricky" feature because, once RPi is shut down,
> you won't be able to do anything remotely through OctoPrint Anywhere until
> you have RPi turned on again.
>
> I usually wait until I'm home to shutdown RPi using OctoPrint directly. Do
> you often run into situation when this is not practical/convenient for you?

Hi Kenneth,

Thanks for replying.  My work cycle is that I have the printer in the
basement and the PC used to design stls and send gcode to the Pi upstairs (
via the Octopi web interface).  So I don't have the web app easy to access
when Im down in the print room.  Usually not a problem but sometimes I need
to kill power when I'm down there but then would have to go up a floor to
do safe shutdown of the pi.   Losing ability to start up remotely is ok for
me since I'll always be powering up when Im down setting up to print anyway.

Not a super big deal (I could bring up the web app on my phone, its just
not very easy to handle on it).

kennethjiang, 2018-07-22:
I see your situation now. 

One of the function on my roadmap for OctoPrint Anywhere is to let users extend its function (for instance, one other user said he'd like to extend it to control the light). It sounds like this can be an easy extension once I have that function built.

Of course, if there are enough users requesting the same feature, I'll be happy to build it.

I'll leave this issue open for a while to see if other users will chime in to make the same request.

xxgithubnz, 2019-08-12:
+1

Also Octoprint Plugin integration would be a very good feature too!

Use case: I would like to trigger GPIO pins on my pi that can turn on/off lights, or can cut power to the printer PSU. This is currently handled by the Enclosure plugin (https://plugins.octoprint.org/plugins/enclosure/), however it would be very handy to be able to do this using OctoPrint Anywhere as the remote is already available and working.

When I'm at home I can monitor my prints via my Android phone using the app Printoid (https://printoid.net). This has the option to manipulate Octoprint plugins that have been made available to the App. eg. the Enclosure plugin above. To be clear, Printoid only supports a few Octoprint plugins so it's not a substitute for the Octoprint interface but it's very handy to have that integration in the App.

Thank you!