kennethjiang/OctoPrint-Anywhere #67: No camera

greg-wood, 2018-10-02:
I am not getting any camera feed either when idle or printing. I am using the multicam plugin, not sure if that might be the issue.

[image]

kennethjiang, 2018-10-02:
It's possible that OA is not compatible with MultiCam plugin. Will investigate it.

kidwon13, 2018-10-03:
I have a similar issue. My OA webcam feed was working and now it's not. I run my webcam on Yawcam connected to a pc. The feed is still working normally in Octoprint, I can access it fine with a browser and I also have the feed set up in Blue Iris and it works okay there too but since yesterday I no longer see it in OA.

kennethjiang, 2018-10-03:
@kidwon13 Are you also using MultiCam plugin?

kidwon13, 2018-10-03:
No I am not. It also started working when I started printing. It just wasn't displaying while idle.

kennethjiang, 2018-10-03:
The cam feed has longer lag when printer is idle to save bandwidth. But if you wait, say more than 30 seconds and the cam feed is still not coming up, then I'm pretty sure it indicates some bug in the code. Will be happy to dig deeper if that's the case.

Can you send me the `octoprint.log` file?

kidwon13, 2018-10-03:
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2442743/octoprint.log)

kennethjiang, 2018-10-03:
Hi @kidwon13 It looks like the wifi network your OctoPi is on has sporadic connectivity issues. For instance, OctoPrint Anywhere couldn't reach to internet during the periods of 2018-10-02 21:34 - 2018-10-02 21:38 (GMT), 2018-10-03 05:30 - 2018-10-03 05:34. There were also times when the Yawcam couldn't be reached.

Next time when you lose the cam feed, can you SSH to your OctoPi to check if it has connectivity to the internet and yamcam server?

kidwon13, 2018-10-03:
I seem to still be losing webcam access when the printer is idle. I did ssh into my raspberry pi and I can ping an internet ip address no problem (I used 4.2.2.2 for my test). Even while I was pinging I did not have the webcam feed on getanywhere.io but I do see it both in the web access to yawcam and on the control tab of octoprint.  I think the wi-fi outages you saw were when I was making configuration changes on my router.  
Curiously I tried to configure a similar setup for my step-dad and he never sees his camera feed on getanywhere.io. It shows his temperatures and status correctly but no webcam feed. I also pinged out from his raspberry pi and it seems to be working okay but he did have a little more packet loss than I did. About 1% over 3 or 4 minutes. 
I'll attach both of our current octoprint.log files. 
[StepDad_octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2444265/StepDad_octoprint.log)
[My_octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2444266/My_octoprint.log)

kennethjiang, 2018-10-03:
Thanks a lot @kidwon13  for the detailed info. It sounds like we need a screen sharing session to debug further.

Can you send me email at kenneth dot jiang at gmail dot com to coordinate a time?

kennethjiang, 2018-10-04:
@greg-wood I installed MultiCam myself and it still works.

@kidwon13 helped me find a bug in the cam feed code (thank you @kidwon13!). Just fixed it and released 0.8.5 for that.

Can you upgrade to 0.8.5 to see if that solves your problem too. If not, can you post the screenshots of your webcam config and MultiCam config page (I posted the screenshots for my configuration as examples)?

[image]
[image]

kennethjiang, 2018-11-05:
Closing it with the assumption that it's fixed.

closed 2018-11-05 by kennethjiang