kennethjiang/OctoPrint-Anywhere #68: I had to remove the plugin because network outages caused Octoprint to crash...

codemonkey2k5, 2018-10-16:
I was beta testing a router that was less then stellar but it brought out a problem with this plug-in.   
If during a print my pi would for any reason lose network connectivity my print would crash.   

And it would do so in a scary way.  The printer would just stop where it was and all heat would stay on.  

For more information and some logs to look at, there is a thread here: https://discourse.octoprint.org/t/caching-print-job-prevent-job-failure-due-to-network-outage/4251

Regards,
Tony

kennethjiang, 2018-10-18:
@codemonkey2k5 Thank you for reporting this issue and sorry for the delay?

By "losing network connectivity", do you mean ethernet connectivity (which seems to be what is indicated in the forum discussion in your link)? I know most of the OA users, myself included are on Wifi, and we lose connectivity quite often but usually it doesn't cause anything except you can't get status update on the dashboard.

I just tested turning off wifi router when it was printing and the print didn't crash. So please let me know if I should focus on testing with ethernet cable (believe or not I don't have one with me right now.)

kennethjiang, 2018-11-05:
Fixed in 0.9.0. Thank you @codemonkey2k5 for testing this fix for me.

closed 2018-11-05 by kennethjiang