kennethjiang/OctoPrint-Anywhere #73: Printer Offline

Foxbiker, 2019-01-13:
Octoprint version 1.3.10 with Klipper SW. Getanywhere page showing video stream but printer is show like offline and remote functions not active.
[image]

kennethjiang, 2019-01-13:
@Foxbiker can you attach the "octoprint.log" file so that I can take a look? Thanks!

Foxbiker, 2019-01-16:
Sorry for delay. Here is the log:
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/2765187/octoprint.log)

kennethjiang, 2019-01-16:
No errors were found in the log file.

Another user had a similar issue recently and it turned out the routing fro his Pi to OA server is somehow not correct. In order to see if you are having the same issue, can you ssh to your Raspberry Pi and run this command:

`time curl https://www.getanywhere.io`

Please post the output of this command here. Thanks!

Foxbiker, 2019-01-16:
[image]

Foxbiker, 2019-01-16:
One additional information. Because I´m using Klipper FW on the printer, Klipper plug-in is installed on the Octoprint. This plug is changing connection panel and maybe can change some statuses in the system. Only my maybe wrong prediction.

kennethjiang, 2019-01-16:
I suspect you are experiencing similar issue with another user. 3.4s is pretty big latency.

I have seen increased latency reports with similar issues very recently. I suspect it's probably caused by some ill-configured router that is acting up and impacting some OA users. I don't think this is something we can do too much about. Hopefully that offending network provider will soon find out and fix the problem. 

Please check OA dashboard regularly to see if things get better or worse. I expect you to experience problems such as not having webcam feed, not having printer status, or both. But I hope that situation will improve within next couple weeks.

Foxbiker, 2019-01-16:
Ok, thank you for advice I will check it.

kennethjiang, 2019-01-17:
Cool thanks!

Foxbiker, 2019-02-05:
Today I try it and it´s still the same - printer ofline, camera working well and remote functions disabled.

Foxbiker, 2019-02-07:
Here is it, I try it 3 times.

[image]

kennethjiang, 2019-02-07:
It looks like the latency for you is consistently long (for most other users it fluctuates a lot).

I'll be migrating to a different hosting service in the next couple weeks. Will ping you again when it's done and test if the problem will be gone with that switch.

kennethjiang, 2019-06-27:
Closing it as there have been very few reports on latency since we moved to a new cloud platform.

closed 2019-06-27 by kennethjiang