kennethjiang/OctoPrint-Anywhere #74: Email sends photo upside down

kjgouldstone, 2019-01-22:
I have rotated the Camera feed using settings however when the photo is sent in the print completion email it is still upside down. The email seems to ignore the settings.

kjgouldstone, 2019-01-22:
Great Thanks

kennethjiang, 2019-01-23:
@kjgouldstone just pushed a version to fix it. Let me know if it works for you.

kjgouldstone, 2019-01-25:
Hi,

I can confirm that the top to bottom flip has worked fine in the fix.

However the left to right flip does not seem to have worked.

See the clock showing 81 rather than 18.

Cheers
[image: unnamed (5).jpg]
Kim

kennethjiang, 2019-01-25:
Ok I will look into it

kennethjiang, 2019-01-25:
What's your printer's orientation settings in OA?

I tested on my printer with only "Flip horizontally" and none of the other options checked. The image in the email came out correctly. I guess you probably also have "rotate 90/90cw" checked? Please let me know so that i can reproduce the problem.

kjgouldstone, 2019-01-25:
Hi,

yes I have flip vertically and flip horizontally ticked

Kim

kennethjiang, 2019-01-25:
Got it fixed. No need to upgrade the plugin in. Your next print should have the correct orientation in email. Thank you for reporting this bug @kjgouldstone

closed 2019-01-25 by kennethjiang

kjgouldstone, 2019-01-28:
Great all working now.

Thanks for the support

Kim