kennethjiang/OctoPrint-Anywhere #76: "error from callback [...]: __on_server_ws_msg__() takes exactly 3 arguments (2 given)"

foosel, 2019-03-28:
I just rolled out 1.3.11rc1 which comes bundled with a new error tracking plugin based on `sentry.io` and I'm seeing a user who's got several instances of 

```
error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x6a462f58>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
```

in their `websocket` log.

Installed version of OctoPrint Anywhere appears to be 0.9.7. 

Can't tell you if it's a general problem or only restricted to this one user, but from the nature of the error it looks like either some old code left over somewhere in the installation or some refactoring mistake or something.

Just wanted to report it before I mark it as "ignored" in my reporting.

foosel, 2019-03-29:
For the record, it's a general issue, overnight the reporting got drowned in these messages from various users and in fact I had to disable error tracking for now as this (and some other third party plugin errors) was eating up the quota on my sponsored account.

tedder, 2019-03-29:
The good news is that it's exposing something is awry.

foosel, 2019-03-29:
True. Also in OctoPrint itself. Found a handful of bugs already, some of which have been in there a while but never were reported.

kennethjiang, 2019-06-27:
This issue should be resolved in version 0.9.10

closed 2019-06-27 by kennethjiang