kennethjiang/OctoPrint-Anywhere #77: "AttributeError: 'AnywherePlugin' object has no attribute 'config'"

foosel, 2019-03-28:
Another thing that popped up thanks to the new error tracking plugin in 1.3.11rc1:

```
AttributeError: 'AnywherePlugin' object has no attribute 'config'
  File "octoprint/server/views.py", line 647, in fetch_template_data
    wizard_required = implementation.is_wizard_required()
  File "__init__.py", line 34, in is_wizard_required
    return not self.config['registered']
```

OctoPrint Anywhere version appears to be 0.9.7 as well.

Just wanted to report it before I mark it as "ignored" in my reporting.

Please tell me if you want to get these kind of things reported or if I should just blindly ignore them ;)

kennethjiang, 2019-03-28:
Thank you for reporting these bugs. @foosel Please keep them coming! :)

kennethjiang, 2019-06-27:
This issue should be already fixed in 0.9.10

closed 2019-06-27 by kennethjiang