kennethjiang/OctoPrint-Anywhere #79: Conflicting with other plugins

owengraven, 2019-04-18:
I have an issue where I can install the plugin once, if I flash a fresh image and then install all the plugins I want, then last octoprint anywhere it works fine after setting it up.

If I add 1 more plugin I can't access the web interface anymore, the only fix is to boot into safemode and disable/uninstall octoprint anywhere.

If I then disable all my plugins, but only leave octoprint anywhere enabled it still doesn't work. Uninstalling and Reinstalling of the plugin also doesn't help.

I suppose it leaves config files and something messes it up.

|      model: Raspberry Pi 3 Model B Rev 1.2
|      octopi_version: 0.16.0
|      octoprint_version: 1.3.10

See included octoprint.log if needed, i coudn't figure out what is wrong, it does throw a warning about something though. 
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3096086/octoprint.log)

owengraven, 2019-04-18:
I also tested this 3 times on fresh installs, and also with only octoprint anywhere installed, in the beginning i didn't know safemode was a thing.

Plugins I use apart from default:
Themify
Navbar Temp
Bed leveling wizard

I don't have any others

kennethjiang, 2019-04-18:
Thank you so much for this detailed bug report @owengraven ! Will look into it!

kennethjiang, 2019-04-18:
Can you also post a screenshot on the page when you can't log into the web interface? I'm trying the same set up and it seems to be working fine.

owengraven, 2019-04-18:
It occurs whenever I install any other plugin after having installed octoprint anywhere.

basically it will try to refresh to reload but fail, and if I close the page and call the ip again it will just keep loading on a blank screen without doing anything. I've read this can occur with pre-emptive caching that it confuses itself. I also tried leaving it for a long time (half an hour) but it still didn't work

kennethjiang, 2019-04-18:
Ok. Will reflash my sd card and test it again from a blank one.

kennethjiang, 2019-04-18:
I still can't reproduce the problem. Here is what I did:

1. Install Octopi 0.16.0 to a formatted sd card.
1. install Themify plugin
1. install navbar temp plugin
1. install OctoPrint Anywhere (skipped registration)
1. install bed leveling wizard plugin (and it still worked)
1. register on OctoPrint Anywhere
1. reboot OctoPi. and it still worked.

I'm on Pi 3 Model B too.
OctoPrint version : 1.3.10
OctoPi version    : 0.16.0

Can you send me an email `k@getanywhere.io` so that we can set up a video chat?  @owengraven 
 This sounds like a tricky issue and I'd like to get to the bottom of it.

owengraven, 2019-04-18:
Tommorow i'll take a full image of my sd card so you can play around with it. I'll do some more testing too.

kennethjiang, 2019-04-18:
That works. I will load your image on my Pi to see what happens.

owengraven, 2019-04-19:
So i made the image, it's hosted on filemail because git didn't let me upload it. 

https://fil.email/j4TAH6PL

Basically it's how i'm running it now, if you install octoprint anywhere it stops working, let me know if I can help troubleshoot more.

kennethjiang, 2019-04-19:
downloading it now. will take a while.

kennethjiang, 2019-04-21:
Thanks again @owengraven for providing me with this image. I found out the problem and fix it in release 0.9.8. Can you reinstall it to see if the problem is fixed?

owengraven, 2019-04-22:
Yep I can confirm the problem is gone. I even tried installing a plugin after it, and deinstall/reinstall and it still worked fine.

Good work mate, thanks for creating this awesome tool.

closed 2019-04-22 by owengraven