kennethjiang/OctoPrint-Anywhere #80: 0.9.8 not updating from 0.9.7

Arakon, 2019-04-22:
When trying to update through Octoprint/Octopi, manually or automatically, the update goes through, no error messages. It then says that the server will restart. After restart, the version remains at 0.9.7 and offers the update again.
I'm running the latest stable Octopi (0.16.0) with Octoprint 1.3.10.

kennethjiang, 2019-04-22:
Thank you so much for reporting the bug. I messed up the versioning in `setup.py`. Just fixed it. Can you try to upgrade one more time to see if it'll still prompt you for the upgrade? @Arakon

dcarusil, 2019-04-22:
Bummer I had the same problem. I tried to re-update twice yesterday but ended up ignoring the prompt afterwards but now it seems that has borked my setup and I have something printing rn lol. Mr arakon please let us know if re-updating works because I will need to

kennethjiang, 2019-04-22:
@dcarusil So OA plugin is not working for you now? You can't see your print status or webcam on OA dashboard?

kennethjiang, 2019-04-22:
It's possible that wrong `setup.py` is cached in OctoPrint. Just pushed 0.9.9 to see if I can overwrite the cache.

dcarusil, 2019-04-22:
@kennethjiang After logging in to OA it shows the dashboard but with the updating icon indefinitely, the little square that flips around. No information is available, I'm flying blind rn

kennethjiang, 2019-04-22:
shit. sorry about that! when you are home and get a chance please update to 0.9.9 and lmk if it doesn't solve it.

Arakon, 2019-04-22:
I just checked and it still tells me 0.9.8 is the latest available update.

kennethjiang, 2019-04-22:
Yeah! Let me know if any of you hit any issues again! Otherwise I will close this issue in a couple days.

dcarusil, 2019-04-23:
@kennethjiang the 0.9.9 update and a reboot fixed the issue! Thanks 😊 Have a good one, happy prints ahead!

closed 2019-05-28 by kennethjiang