kennethjiang/OctoPrint-Anywhere #81: No camera feed, everything else works

FledPotato, 2019-04-29:
Using windows to run octoprint on a laptop and octoprint anywhere cant seem to get video. I still get it on the regular octoprint site, and controls/reads work great on both.

kennethjiang, 2019-04-29:
Can you post your `octoprint.log` file so I can take a look?

FledPotato, 2019-04-29:
Am I able to do that while a print is going? If not itll be done in about 20 minutes, then I can send you all the info and follow whatever steps you need to see the issue

kennethjiang, 2019-04-29:
I think it's ok to download the log files while it's printing.

FledPotato, 2019-04-29:
[octoprint.log](https://github.com/kennethjiang/OctoPrint-Anywhere/files/3129093/octoprint.log)
Is this right? Im on my phone haha, also it says the file is from yesterday but I guess thats fine? If theres just a mess there i can try to get a new log

kennethjiang, 2019-04-29:
Thanks @FledPotato 

From the log file it looks like the plugin couldn't connect to the configured webcam URL. Can you post a screenshot on your webcam and timelapse settings? [This link](https://www.thespaghettidetective.com/docs/webcam-feed-is-not-showing/#step-3-make-sure-webcam-url-settings-are-correct-in-octoprint) is for my another plugin but is relevant for the problem you are experiencing.

FledPotato, 2019-04-29:
[image]
The timelapse link thing doesnt work, but normal webcam stream does, also can i just show my ip to everyone like this?

kennethjiang, 2019-04-29:
I'm not a security expert but since it's your private ip I'd assume it's fine. I've been doing it many times.

Can you remove the value from `snapshot URL`? The plugin code looks for it first, and will try to use it if it's there. After you remove it, the plugin will fall back to webcam URL after OctoPrint is restarted.

FledPotato, 2019-04-29:
Still "waiting for camera feed"

FledPotato, 2019-04-30:
As well as i noticed the printer status is "idle" even if im mid print

kennethjiang, 2019-05-05:
Sorry somehow I missed your comments.

Did you clear the `snapshot URL` in your OctoPrint settings?

It sounds like a bigger problem if the printer status is off. In that case, you can download `octoprint.log` file and post it here again?

FledPotato, 2019-05-05:
Im mid print right now without ocotprint running, could I do this tomorrow? (11pm here) Then i can do all the troubleshooting after a fresh start of octoprint so the log isnt as messy, if thats how it works

kennethjiang, 2019-05-05:
A "messy" log is fine. The errors usually hide in all this messiness. :)

FledPotato, 2019-05-10:
Sorry for the late response, but seems the issue was fault of another program, which was streaming my webcam. Switched software and reinstalled, now it works great! Is there any chance I could have a higher streaming fos though? My laptop is whats running ocotoprint so the webcam is always at 20-30 fps, which is quite a bit more than the capped site shows :/

kennethjiang, 2019-05-10:
Glad you figured this out!

We are actively working on a premium webcam streaming service that allows user to go to higher frame rate. However it'll be a paid service as it costs a lot to transfer the webcam pictures between user's network and our servers.

closed 2019-05-22 by kennethjiang