kennethjiang/OctoPrint-Anywhere #82: Time lapses no longer recording or uploading

ShibumiDesign, 2019-05-03:
Having an issue where I can monitor and basic control my printer, but when starting a job through octoprint it only records time lapses natively on the octopi and does not record or upload through octoprint anywhere... nor does it email me when the print job is complete. I did however install custom marlin firmware that adds semi automatic mesh bed levelling. Last working time lapse through octoprint anywhere was April 20th.

Did the custom marlin firmware break octoprint anywhere plugin? Like I said, I can still see live feed of my printer and basic movement controls, pause/cancel an active print, just no time lapses anymore or email notification. I’ve tried turning off and on again the record time lapse and upload time lapses, but still no time lapses show after April 20th.

Any help greatly appreciated.

kennethjiang, 2019-05-05:
@ShibumiDesign sorry for the issue you are experiencing.

If you add `support@getanywhere.io` to your address book the emails should stop going to spam folder.

Do you have Octolapse plugin installed and enabled? If so you want to toggle on both `Record timelapse for you?` and `Upload Octolapse timelapses?`, as in this screenshot.

[image]

However, if you don't have Octolapse installed or have it disabled, you should only toggle on `Record timelapse for you?`. `Upload Octolapse timelapses?` should be toggled *OFF* otherwise the timelapses won't generated.

Let me know if your settings are correct but still not getting timelapses generated.

ShibumiDesign, 2019-05-05:
Hi, I actually just added octolapse to my system, and now they are uploading... octolapse giving me much nicer time lapses than the original.

Thanks for the reply

kennethjiang, 2019-05-06:
Cool! Closing the ticket.