kennethjiang/OctoPrint-Anywhere #85: Webcam framerate is much slower than on LAN

Mattwen, 2019-05-29:
The frame rate on OctoPrint-Anywhere is much slower than the frame rate of my Webcam on my LAN. Is there any way to fix this?

kennethjiang, 2019-05-30:
OA streams webcam feed through the cloud (this is how you can access OctoPrint without port-forwarding), so we have to keep the frame rate low otherwise it'll cost too much internet bandwidth for most users.

However, we will be releasing a highly efficient video streaming (the same way as how YouTube streams videos) in a few weeks, which will give you 24 fps with about the same internet bandwidth requirement as current webcam feed. It'll cost $5/month to be eligible for this new video streaming (because processing video streaming in cloud is expensive). Stay tuned.

kennethjiang, 2019-06-27:
Closing it since 25 fps streaming is officially available to everyone in OctoPrint Anywhere 1.0

closed 2019-06-27 by kennethjiang